usfca-cs-tools/golden-gates #37: GGL: more pythonic syntax

phpeterson-usf, 2025-06-02:
Mehmet suggests:

And(a, b, c)
and1 = And(a, b, c)
and1.input[0]
and1.input[0] <<= a

phpeterson-usf, 2025-06-02:
I think using <<= (i.e. override __lshift__) would be easy enough. Seems like it could just be syntactic sugar around Circuit.connect(). However, last time I talked to @gdbenson about operator overloads, he wasn't a fan.

I did the inputs/outputs as strings (and1.inputs['0']) because
1. it seemed like a straightforward way to write the data structure for inputs and outputs
2. it seemed extensible into named inputs/outputs like
```python
c = circuit.Circuit()
m = plexers.Multiplexer(bits=3)
i = io.Input(bits=3)
c.connect(i, m.inputs['sel'])
```

but maybe the __lshift__ thing covers that:
```python
m.sel <<= i
```

maemre, 2025-06-03:
Another option is to override `__setitem__` in `NodeInputs`, it already [overrides `__getitem__`](https://github.com/phpeterson-usf/golden-gates/blob/main/web/public/ggl/node.py#L33) for the `node.inputs[i]` syntax. This would allow

```
and1.input[0] = a
```

The only issue is getting the ambient circuit to be able to call `c.connect`.

This would limit assignments to only inputs/outputs of nodes but it is less "magic". If we want to connect two arbitrary connectors, then overriding `Connector.__ilshift__` would be more flexible (e.g., it'd allow `m.sel <<= i`).