usfca-cs-tools/golden-gates #70: Offer to auto-connect wires

phpeterson-usf, 2025-07-17:
1. User inserts a logic gate
2. User inserts an input near (facing?) the input to a gate 
3. User inserts an output near (facing?) the output to a gate
4. GG draws a dotted line to propose a wire, along with some yes/no UI affordance for the user to accept the wire

Ndewedo-Newbury, 2026-10-10:
I could work on some of these issues with claude if you'd like?

Ndewedo-Newbury, 2026-10-21:
@phpeterson-usf could you review this before I start building? I'd like a human sign-off on the design, especially the trigger and matching rules, before any code goes in.

**Proposed design for auto-connect suggestions**

**When it runs**
- After placing a component, from a sidebar drag or a menu/shortcut insert.
- After dragging a single existing component. Multi-selection drags don't trigger it.
- It applies to any component type with ports, not only gates and inputs/outputs.

**Which pairs qualify**
- One output port and one input port.
- The two ports face in opposite directions. Each port's direction comes from its component's rotation.
- Each port lies in front of the other.
- They are at most 4 grid units apart (Manhattan distance, about 60px).
- The input port has no wire on it yet. Output ports can already have wires, since fan-out is allowed.
- The bit widths match. If a port's width can't be worked out, it matches any width.

**If there are several candidates**
- Each free input gets at most one proposal: its nearest eligible output.
- Ties go to the straight wire, then to the lower y.
- All proposals from one placement are offered together.

**What the user sees**
- Proposed wires are drawn as dashed orthogonal wires.
- A small ✓/✕ chip sits next to the proposals.
- Enter accepts all of them and Esc rejects all of them.
- Proposals stay when the user pans or zooms.
- Any edit, or a click on empty canvas, quietly dismisses them.

**Undo**
- Accepting is its own undo step, separate from the placement. One Ctrl+Z removes the accepted wires and keeps the component.

**Setting**
- The feature is on by default.
- A toggle stores the choice in localStorage under `gg.autoConnectSuggestions`.
- All new UI text goes through the locale files (i18n).

**Planned PRs**
1. `utils/autoConnect.js`: port facing, port widths and the matching logic, with unit tests. No UI changes.
2. The proposal display, the chip, the Enter/Esc keys, dismissal, undo, the toggle, the i18n strings, and integration tests.

phpeterson-usf, 2026-10-21:
Looks solid! Are there test cases for it? Could you add it on a branch so we can try it out?